8devices/carambola2 #50: Supported libraries - glib2 dbus1

JacekOort, 2015-02-02:
Dear Sir/Madam

I am trying to configure new linux image and sysroot for Carambola2. I cloned newest Carambola sources from git repository and I have version v2.6. Unfortunately in menuconfig I can not find libraries that are necessary for our project. I mean:
 *glib-2.0
 *dbus-1

As I noticed, in previous version of carambola software, in some Makefiles I can see glib2 in dependencies. Was that libraries removed form project for some reason or it is possible to unlock this libraries in menuconfig?

mantas-p, 2015-02-02:
Did you follow steps at http://8devices.com/wiki/carambola:2:gettingstarted section 'Build firmware'?
Both glib2 and dbus are available in menuconfig. Check if you pulled all package feeds correctly.

mantas-p, 2015-03-04:
Assume problem was resolved, reopen if not.

closed 2015-03-04 by mantas-p