8devices/carambola2 #53: compilation fails with 404 not found ( apr-util)

shroukkhan, 2015-02-22:
hi, 
while trying to compile, it is failing with the following error:

make[3]: Leaving directory `/home/khan/compilations/carambola2.frcd/feeds/packages/libs/sqlite3'
make[3]: Entering directory `/home/khan/compilations/carambola2.frcd/feeds/packages/libs/apr-util'
mkdir -p /home/khan/compilations/carambola2.frcd/dl
/home/khan/compilations/carambola2.frcd/scripts/download.pl "/home/khan/compilations/carambola2.frcd/dl" "apr-util-1.5.3.tar.bz2" "6f3417691c7a27090f36e7cf4d94b36e" "http://mirrors.ibiblio.org/apache/apr/"
--2015-02-22 05:55:53--  http://mirrors.ibiblio.org/apache/apr/apr-util-1.5.3.tar.bz2
Resolving mirrors.ibiblio.org (mirrors.ibiblio.org)... 152.19.134.44
Connecting to mirrors.ibiblio.org (mirrors.ibiblio.org)|152.19.134.44|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-02-22 05:55:54 ERROR 404: Not Found.

Download failed.
--2015-02-22 05:55:54--  http://mirror2.openwrt.org/sources/apr-util-1.5.3.tar.bz2
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-02-22 05:55:55 ERROR 404: Not Found.

Download failed.
--2015-02-22 05:55:55--  http://downloads.openwrt.org/sources/apr-util-1.5.3.tar.bz2
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-02-22 05:55:57 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
make[3]: *** [/home/khan/compilations/carambola2.frcd/dl/apr-util-1.5.3.tar.bz2] Error 2
make[3]: Leaving directory `/home/khan/compilations/carambola2.frcd/feeds/packages/libs/apr-util'
make[2]: *** [package/feeds/packages/apr-util/compile] Error 2
make[2]: Leaving directory `/home/khan/compilations/carambola2.frcd'
make[1]: *** [/home/khan/compilations/carambola2.frcd/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/khan/compilations/carambola2.frcd'
make: *** [world] Error 2

it seems it is trying to download apr-util-1.5.3.tar.bz2 which does not exist anymore. this was compiling fine last week.

i have tried this:

./scripts/feeds update -a
./scripts/feeds install -a

to make sure i have the latest updates..still no good.

any ideas?

                    

                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                       

GiedriusM, 2015-02-22:
The builder fails to download sources for this package, because it does not exist anymore. It looks like the package was updated to 1.5.4 version and the older sources were removed. There are a few things that you can try:
1) Update package information so that it would use the newer version
or 
2) Dowload the apr-util-1.5.3.tar.bz2 manually from the archives (https://archive.apache.org/dist/apr/apr-util-1.5.3.tar.bz2), put it into dl folder and re-run the make command.

shroukkhan, 2015-02-22:
downloading manually from archive did it ( suggestion 2 ).

however, 

./scripts/feeds update -a
./scripts/feeds install -a

i did try those ^ . but it didnt fix anything. isnt that how the package information is updated?

GiedriusM, 2015-02-22:
Yes, the above commands install the package information (like what packages are available, names, descriptions, etc.), but not the actual packages. When you're building Carambola2 image, the build scripts first check if it has the sources of the selected packages and only then starts compiling them. If it doesn't (for example when it's a first build, or new packages were selected in menuconfig), then it downloads and extracts the package source code.
If all of the AVAILABLE package source code downloaded as opposed to only the SELECTED ones, then it would consume (waste) a lot of disk space.
I hope this helps and if it solved your problem and answered your question, please close the issue.

closed 2015-03-04 by GiedriusM