8devices/carambola2 #54: Compile shared openssl 1.0.1j

OortJacek, 2015-05-19:
Hi, 
We want to add openssl in version 1.0.1j to compile lib mosquitto. Now I can only compile openssl in static library version. How to do this with shared library?

//------------------------------
Configured for os/compiler.

You gave the option 'shared'.  Normally, that would give you shared libraries.
Unfortunately, the OpenSSL configuration doesn't include shared library support
for this platform yet, so it will pretend you gave the option 'no-shared'.  If
you can inform the developpers (openssl-dev\@openssl.org) how to support shared
libraries on this platform, they will at least look at it and try their best
(but please first make sure you have tried with a current version of OpenSSL).
//----------------------------

Regards
Jacek

GiedriusM, 2015-05-19:
I think the quotation answers your question.

OortJacek, 2015-05-19:
But what kind of os/compiler did You pick and what kind of openssl wersion did You compile?

I am trying with oort@oort-jacek:~/carambola/openssl$ ./config shared --openssldir=$SDKTARGETSYSROOT/usr  os/compiler:linux-mipsel

OortJacek, 2015-05-19:
... and linux-mips

GiedriusM, 2015-05-19:
The available ssl options can be found in Libraries->SSL in menuconfig. Also you should add packages into the openwrt build system rather than trying to install them separately.

GiedriusM, 2015-05-19:
Plus there's also a libmosquitto libraries (both ssl and non-ssl versions) available in the Library menu.

OortJacek, 2015-05-25:
Ok, I added libmosquitoo from menuconfig but I think that dependencies are still unresolved. I can link with libmosquito only with warrning:

mips-openwrt-linux-uclibc/bin/ld: warning: librt.so.0, needed by /home/oort/carambola/carambola2/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib/libmosquitto.so, not found (try using -rpath or -rpath-link)

mantas-p, 2015-05-25:
Do you get this warning when you compile libmosquitto or your application?
Does compilation succeed despite the warning?

OortJacek, 2015-05-26:
Warning occurs when I am linking my applications. Compilation succeed but I got this kind of errors in my app:

OpenSSL Error: error:100AE081:lib(16):func(174):reason(129)
OpenSSL Error: error:1408D010:lib(20):func(141):reason(16)

that pop out from mosqquito and I am unable to decode.

OortJacek, 2015-05-26:
I researched something and I found that might be included with ECC compare error code with someone who has this error code and error string :
http://stackoverflow.com/questions/19503174/connecting-to-xmpp-with-ssl-throws-exception-error100ae081elliptic-curve-rout

The problem is described here:
https://rhn.redhat.com/errata/RHBA-2013-1751.html 

Do You agree? How to fix this?

valinskas, 2015-05-27:
It looks like this code 

```C
3129 EC_GROUP *EC_GROUP_new_by_curve_name(int nid)
3130 {
3131     size_t i;
3132     EC_GROUP *ret = NULL;
3133
3134     if (nid <= 0)
3135         return NULL;
3136
3137     for (i = 0; i < curve_list_length; i++)
3138         if (curve_list[i].nid == nid) {
3139             ret = ec_group_new_from_data(curve_list[i]);
3140             break;
3141         }
3142
3143     if (ret == NULL) {
3144         ECerr(EC_F_EC_GROUP_NEW_BY_CURVE_NAME, EC_R_UNKNOWN_GROUP);
3145         return NULL;
3146     }
```

ec_group_new_from_data() cannot find ECC by given ID and returns NULL, line 3144. It seems to me you can try fixing openssl or reconfigure mqtt server so it uses other ECC (instead of unknown or swich from ECC to some other algo on server).

closed 2021-09-27 by valinskas