8devices/carambola2 #55: Remote debug with GDB and Barrier braker

Darlard, 2015-05-25:
Hello,
with the new version of the kernel i'm not able to debug anymore. Everytime GDB hits a libc functions, the disassebler crashes with SIGILL or SIGSEGV. I can't step over library functions even if the whole kernel is compiled with Global debug, and uClibc is compiled with debug symbols.

it always fail on jal instructions to .plt section, like <strerror@.plt> or <printf@.plt>
Command "info sharedlibrary" doesn't show missing symbols, and apparently i'm able to retrieve those symbols addresses and vice versa with gdb "info" command.

With previous version of kernel everything was working fine. I also tried to compile official source code and i get same errors. 

Any idea why i am failing in remote debugging?
Thanks

mantas-p, 2015-05-25:
Hi,

Are you running gdbserver on device, or use JTAG as debug interface?
I don't have any ideas what may be causing this. If it happens with official OpenWRT builds too, ask in their forums.
Only way to debug I can think of would be to basically git-bisect between working and non-working versions. Too time consuming I guess.

Darlard, 2015-05-25:
Hi,
i'm running gdbserver on device. I did ask in OpenWRT forum, but got no answer.
I was able to remote debug with your 2.6 source, but with 2.7 i can't anymore.

valinskas, 2021-09-27:
Closing issue, 5y+ since last activity.

closed 2021-09-27 by valinskas