8devices/carambola2 #58: Building - recipe for target 'world' failed

1am, 2015-09-28:
Hi,

I've been trying to build the Carambola image but for some vague reason it fails. First time it failed it was due to insufficent disk space on the VM but after increasing it still fails with the "*recipe for target 'world' failed*" exit message. Does someone has had a simmilar issue? It repeats every time even though I've removed the *build_dir* and *staging_dir*. From what I see the ```CONFIG_PACKAGE_libdbus=y
``` is set by default in the *.config* file.

The `make V=s` logs from just before failing are pasted below:
```
...
OpenWrt-libtool: link: mv -f ".libs/libdbus-1.expT" ".libs/libdbus-1.exp"
OpenWrt-libtool: link: echo "{ global:" > .libs/libdbus-1.ver
OpenWrt-libtool: link:  cat .libs/libdbus-1.exp | sed -e "s/\(.*\)/\1;/" >> .libs/libdbus-1.ver
OpenWrt-libtool: link:  echo "local: *; };" >> .libs/libdbus-1.ver
OpenWrt-libtool: link:  mips-openwrt-linux-uclibc-gcc -shared  -fPIC -DPIC  .libs/libdbus_1_la-dbus-address.o .libs/libdbus_1_la-dbus-auth.o .libs/libdbus_1_la-dbus-bus.o .libs/libdbus_1_la-dbus-connection.o .libs/libdbus_1_la-dbus-credentials.o .libs/libdbus_1_la-dbus-errors.o .libs/libdbus_1_la-dbus-keyring.o .libs/libdbus_1_la-dbus-marshal-header.o .libs/libdbus_1_la-dbus-marshal-byteswap.o .libs/libdbus_1_la-dbus-marshal-recursive.o .libs/libdbus_1_la-dbus-marshal-validate.o .libs/libdbus_1_la-dbus-message.o .libs/libdbus_1_la-dbus-misc.o .libs/libdbus_1_la-dbus-nonce.o .libs/libdbus_1_la-dbus-object-tree.o .libs/libdbus_1_la-dbus-pending-call.o .libs/libdbus_1_la-dbus-resources.o .libs/libdbus_1_la-dbus-server.o .libs/libdbus_1_la-dbus-server-debug-pipe.o .libs/libdbus_1_la-dbus-server-socket.o .libs/libdbus_1_la-dbus-uuidgen.o .libs/libdbus_1_la-dbus-server-unix.o .libs/libdbus_1_la-dbus-sha.o .libs/libdbus_1_la-dbus-signature.o .libs/libdbus_1_la-dbus-syntax.o .libs/libdbus_1_la-dbus-timeout.o .libs/libdbus_1_la-dbus-threads.o .libs/libdbus_1_la-dbus-transport.o .libs/libdbus_1_la-dbus-transport-socket.o .libs/libdbus_1_la-dbus-watch.o .libs/libdbus_1_la-dbus-dataslot.o .libs/libdbus_1_la-dbus-file.o .libs/libdbus_1_la-dbus-hash.o .libs/libdbus_1_la-dbus-internals.o .libs/libdbus_1_la-dbus-list.o .libs/libdbus_1_la-dbus-marshal-basic.o .libs/libdbus_1_la-dbus-memory.o .libs/libdbus_1_la-dbus-mempool.o .libs/libdbus_1_la-dbus-pipe.o .libs/libdbus_1_la-dbus-string.o .libs/libdbus_1_la-dbus-file-unix.o .libs/libdbus_1_la-dbus-pipe-unix.o .libs/libdbus_1_la-dbus-sysdeps-unix.o .libs/libdbus_1_la-dbus-sysdeps-pthread.o .libs/libdbus_1_la-dbus-transport-unix.o .libs/libdbus_1_la-dbus-userdb.o .libs/libdbus_1_la-dbus-sysdeps.o   -L/home/pb/Desktop/carambola/carambola2/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib -L/home/pb/Desktop/carambola/carambola2/staging_dir/target-mips_34kc_uClibc-0.9.33.2/lib -L/home/pb/Desktop/carambola/carambola2/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/pb/Desktop/carambola/carambola2/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib -lpthread  -Os -mno-branch-likely -mips32r2 -mtune=34kc -msoft-float -mips16 -minterlink-mips16 -Wl,-rpath-link=/home/pb/Desktop/carambola/carambola2/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib   -Wl,-soname -Wl,libdbus-1.so.3 -Wl,-version-script -Wl,.libs/libdbus-1.ver -o .libs/libdbus-1.so.3.11.3
/home/pb/Desktop/carambola/carambola2/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib/gcc/mips-openwrt-linux-uclibc/4.8.3/../../../../mips-openwrt-linux-uclibc/bin/ld:.libs/libdbus-1.ver:2: syntax error in VERSION script
collect2: error: ld returned 1 exit status
Makefile:1025: recipe for target 'libdbus-1.la' failed
make[7]: *** [libdbus-1.la] Error 1
make[7]: Leaving directory '/home/pb/Desktop/carambola/carambola2/build_dir/target-mips_34kc_uClibc-0.9.33.2/dbus-1.9.10/dbus'
Makefile:940: recipe for target 'all' failed
make[6]: *** [all] Error 2
make[6]: Leaving directory '/home/pb/Desktop/carambola/carambola2/build_dir/target-mips_34kc_uClibc-0.9.33.2/dbus-1.9.10/dbus'
Makefile:596: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/pb/Desktop/carambola/carambola2/build_dir/target-mips_34kc_uClibc-0.9.33.2/dbus-1.9.10'
Makefile:500: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/pb/Desktop/carambola/carambola2/build_dir/target-mips_34kc_uClibc-0.9.33.2/dbus-1.9.10'
Makefile:211: recipe for target '/home/pb/Desktop/carambola/carambola2/build_dir/target-mips_34kc_uClibc-0.9.33.2/dbus-1.9.10/.built' failed
make[3]: *** [/home/pb/Desktop/carambola/carambola2/build_dir/target-mips_34kc_uClibc-0.9.33.2/dbus-1.9.10/.built] Error 2
make[3]: Leaving directory '/home/pb/Desktop/carambola/carambola2/feeds/packages/utils/dbus'
package/Makefile:174: recipe for target 'package/feeds/packages/dbus/compile' failed
make[2]: *** [package/feeds/packages/dbus/compile] Error 2
make[2]: Leaving directory '/home/pb/Desktop/carambola/carambola2'
package/Makefile:171: recipe for target '/home/pb/Desktop/carambola/carambola2/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/home/pb/Desktop/carambola/carambola2/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/pb/Desktop/carambola/carambola2'
/home/pb/Desktop/carambola/carambola2/include/toplevel.mk:179: recipe for target 'world' failed
make: *** [world] Error 2

```

mantas-p, 2015-09-28:
Hi,

have you tried running ```make dirclean``` after failed build?
I've just did a clean build and it finished successfully. If dirclean doesn't help try also:
```
make distclean
git checkout .config
make menuconfig
make V=s -j4
```

1am, 2015-09-28:
Thank you. ```make distclean``` did the job.

closed 2015-09-28 by 1am

jtelocken, 2017-09-29:
Hello everyone, 
I have the same issue.
Has anyone been able to solve this case? I'm using OpenSuse Leap 42.3. I've tried practically all the tips found in Web, about this problem and I can not solve it. I'll be grateful for any tips.

________________________________________________________________________________________________________
 make[1] world
 make[2] tools/install
 make[2] package/cleanup
 make[3] -C tools/flock compile
 make[3] -C tools/flock install
 make[3] -C tools/tar compile
make -r world: build failed. Please re-run make with -j1 V=s to see what's going on
/home/jonas/buildroot_desenv/3buildroot_novo/**carambola2/include/toplevel.mk:192: recipe for target 'world' failed**
make: *** [world] Error 1
________________________________________________________________________________________________________

Thanks for now.

tomwuxj, 2019-12-23:
> Hi,
> 
> have you tried running `make dirclean` after failed build?
> I've just did a clean build and it finished successfully. If dirclean doesn't help try also:
> 
> ```
> make distclean
> git checkout .config
> make menuconfig
> make V=s -j4
> ```

it helps me also. What is the magic?