8devices/carambola2 #59: Extroot overlay parition

1am, 2015-10-05:
Hi,

I've posted a question regarding configuring Extroot on Carambola2 a few days ago and I thought that maybe someone here would know he answer to my problems? To not duplicate content I'm pasting a link to the original question: http://www.8devices.com/community/viewtopic.php?f=13&t=914

I'm a bit stuck and don't really know what I'm doing wrong and if there's something specific to Carambola OpenWRT distribution which I have to consider when configuring extroot?

mantas-p, 2015-10-08:
Hi,

It seems there is a bug in current Carambola OpenWRT distribution when handling extroot. It will be fixed for next release. For now I made it work with the following workaround:

Download these packages from OpenWRT:
```
cd /tmp
wget http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/packages/base/libubox_2015-06-14-d1c66ef1131d14f0ed197b368d03f71b964e45f8_ar71xx.ipk
wget http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/packages/base/block-mount_2015-05-24-09027fc86babc3986027a0e677aca1b6999a9e14_ar71xx.ipk
```

Install them
```
opkg install block-mount_2015-05-24-09027fc86babc3986027a0e677aca1b6999a9e14_ar71xx.ipk 
opkg install libubox_2015-06-14-d1c66ef1131d14f0ed197b368d03f71b964e45f8_ar71xx.ipk
```

Make sure /etc/rc.local contains this line:
```
export PREINIT=1;mount_root
```

After this extroot mounted successfully for me:
```
Filesystem                Size      Used Available Use% Mounted on
rootfs                    5.8G     12.9M      5.5G   0% /
/dev/root                 3.0M      3.0M         0 100% /rom/rom
tmpfs                    29.9M    336.0K     29.6M   1% /tmp
/dev/mtdblock5           11.8M    816.0K     11.0M   7% /rom/overlay
overlayfs:/overlay       11.8M    816.0K     11.0M   7% /rom
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/sda1                 5.8G     12.9M      5.5G   0% /overlay
overlayfs:/overlay        5.8G     12.9M      5.5G   0% /

rootfs on / type rootfs (rw)
/dev/root on /rom/rom type squashfs (ro,relatime)
proc on /proc type proc (rw,noatime)
sysfs on /sys type sysfs (rw,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/mtdblock5 on /rom/overlay type jffs2 (rw,noatime)
overlayfs:/overlay on /rom type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
tmpfs on /dev type tmpfs (rw,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,relatime,mode=600)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
/dev/sda1 on /overlay type ext4 (rw,relatime,data=ordered)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
```

1am, 2015-10-08:
Thank you! It worked perfectly

closed 2015-10-09 by mantas-p