8devices/carambola2 #60: Customizing files when building sysupgrade image

1am, 2015-12-05:
Hello,

I wanted to ask about customization of the whole filesystem which will be created on a Carambola2 based device after re-flashing. 

Long story short, I've made a prototype which is working well and wanted to transfer these settings to more devices, preferably in some automated way... From what I've found so far it's not possible to just make a complete copy of a working system and transfer it to another device. So right now I'm configuring the build process to create the sysupdate image which I need (with all the network, USB devices configured like they should) and I'm kind of stuck with the root filesystem.

After some more reading I've found that by default ImageBuilder will look for the file system structure in *<buildroot dir>/files/* [via OpenWRT wiki](https://wiki.openwrt.org/doc/howto/build?s#custom_files). The thing is that I can find multiple *files* subdirectories scattered all over the directories from the repo and build dirs, none of which resemble the final structure - some are missing */etc/config/network* files while others contain only the file... 
Looking a bit further I've decided to take a look at [Arduino Yun ditribution repo](https://github.com/arduino/openwrt-yun) which already contains the directory I'm looking for.

Given that I've made a bold assumption that the Carambola makefile was modified in that part to look elsewhere for the root file structure... So given all that, my humble question would be: how do I add/remove/edit file structure which will later be compiled into the *openwrt-ar71xx-generic-carambola2-squashfs-sysupgrade.bin*?

mantas-p, 2015-12-07:
Hi,

Every package in OpenWRT can supply files/ directory for install to file-system, so normally rootfs contents will depend on packages you select. In your case, you can just put your customized files to <buildroot dir>/files/ and it will override any files from packages.

1am, 2015-12-07:
Thank you! I'll try it out later today. 
What would be your suggested approach to have some unique per-device settings bundled with the image (eg. password)?

mantas-p, 2015-12-07:
Well you can't have per-device settings in image, unless you generate separate image for every device. You can generate some settings with scripts on first boot (/etc/config/network is generated that way). MAC address is the only unique information per-device after flashing, but it's not good for generating passwords if you want them unpredictable.

1am, 2015-12-07:
Just to clarify, the `buildroot` directory is the root dir of the repository or `build_dir` which is created during the build process?

Where could I find the script for the /etc/config/network generation? I'm not planning to generate passwords with this but it might be good reference for learning how things are set up.

mantas-p, 2015-12-07:
```<buildroot dir>``` is root dir of repo (it's called ```carambola2``` for this repo).

This script generates network config based on board info:

``` target/linux/ar71xx/base-files/etc/uci-defaults/02_network ```

1am, 2015-12-07:
Thank you! I've tested with custom /etc/banner and it worked.

closed 2015-12-07 by 1am

1am, 2015-12-08:
If I'd want to add custom scripts executed like `target/linux/ar71xx/base-files/etc/uci-defaults/02_network` do they need to be placed in the same path (`target/linux/ar71xx/base-files/etc/uci-defaults/`) to be executed, or there's also some place for custom first boot?

Maybe there is some reference on the particular directories of the OpenWRT build repo / scripts? I can't find mich in the [OpenWrt build system wiki](https://wiki.openwrt.org/doc/howto/build)

mantas-p, 2015-12-10:
You can place them either in ```target/linux/ar71xx/base-files/etc/uci-defaults/``` or ```files/etc/uci-defaults/```, they will end up in same place anyway.