8devices/carambola2 #62: firmware build failed (undefined reference to `__atomic_load_8' )

s-light, 2016-02-04:
i tried to build the firmware following  [Getting Started - Build firmware](http://www.8devices.com/wiki/carambola:2:gettingstarted).
this was buggy:
```
...
mips_34kc_musl-1.1.11/usr/lib -L/home/stefan/carambola2/staging_dir/target-mips_34kc_musl-1.1.11/lib -L/home/stefan/carambola2/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/usr/lib -L/home/stefan/carambola2/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/lib lib/.libs/libopenvswitch.so
lib/.libs/libopenvswitch.so: undefined reference to `__atomic_load_8'
lib/.libs/libopenvswitch.so: undefined reference to `__atomic_store_8'
collect2: error: ld returned 1 exit status
...
```
i had to add `OVS_CHECK_ATOMIC_LIBS` to
`carambola2/build_dir/target-mips_34kc_musl-1.1.11/openvswitch-2.4.0/configure.ac`
i posted a [detailed error log in the forum](http://www.8devices.com/community/viewtopic.php?f=13&t=925). And stogey25 found this solution.

i don't know if this is carambola specific.

mantas-p, 2016-02-04:
Hi,

Build process changed a bit but documentation didn't :| . There's ```build.sh``` script in root directory for simple building.
If you do repeated builds just do:
```
make -j4 IGNORE_ERRORS=m
```
To skip broken packages

valinskas, 2021-09-27:
Closing issue, 4y+ since last activity.

closed 2021-09-27 by valinskas