8devices/carambola2 #63: `opkg update` results in `signature check failed`

s-light, 2016-02-04:
i build the firmware following the [Getting Started - Build firmware](http://www.8devices.com/wiki/carambola:2:gettingstarted).
updated with the Web interface / LUCI method.
after this i tried a `opkg update`
but that failed with `signature check failed.`
```
root@Carambola2:/# opkg update
   Downloading http://pkg.8devices.com/carambola2/v2.8/packages/base/Packages.gz.
   Updated list of available packages in /var/opkg-lists/designated_driver_base.
   Downloading http://pkg.8devices.com/carambola2/v2.8/packages/base/Packages.sig.
   Signature check failed.
   Remove wrong Signature file.
   Downloading http://pkg.8devices.com/carambola2/v2.8/packages/telephony/Packages.gz.
   Updated list of available packages in /var/opkg-lists/designated_driver_telephony.
   Downloading http://pkg.8devices.com/carambola2/v2.8/packages/telephony/Packages.sig.
   Signature check failed.
   Remove wrong Signature file.
   Downloading http://pkg.8devices.com/carambola2/v2.8/packages/packages/Packages.gz.
   Updated list of available packages in /var/opkg-lists/designated_driver_packages.
   Downloading http://pkg.8devices.com/carambola2/v2.8/packages/packages/Packages.sig.
   Signature check failed.
   Remove wrong Signature file.
   Downloading http://pkg.8devices.com/carambola2/v2.8/packages/routing/Packages.gz.
   Updated list of available packages in /var/opkg-lists/designated_driver_routing.
   Downloading http://pkg.8devices.com/carambola2/v2.8/packages/routing/Packages.sig.
   Signature check failed.
   Remove wrong Signature file.
   Downloading http://pkg.8devices.com/carambola2/v2.8/packages/luci/Packages.gz.
   Updated list of available packages in /var/opkg-lists/designated_driver_luci.
   Downloading http://pkg.8devices.com/carambola2/v2.8/packages/luci/Packages.sig.
   Signature check failed.
   Remove wrong Signature file.
   Downloading http://pkg.8devices.com/carambola2/v2.8/packages/management/Packages.gz.
   Updated list of available packages in /var/opkg-lists/designated_driver_management.
   Downloading http://pkg.8devices.com/carambola2/v2.8/packages/management/Packages.sig.
   Signature check failed.
   Remove wrong Signature file.
```
how can i fix this?
first i asked [this question in the forum](http://www.8devices.com/community/viewtopic.php?f=13&t=927)

mantas-p, 2016-02-04:
Hi,

Since you built firmware yourself, it generated new repository key. To use packages from pkg.8devices.com with this firmware you can:

Add 8devices repo key to your device
```
echo "untrusted comment: Local build key
RWRIT89nf01hfEZ0sl+4YABtaeWs38ms/ZwOxayAQtW8Ilxt74oZWQEH" > /etc/opkg/keys/484fcf677f4d617c
```
--OR--

Disable signature verification in ```/etc/opkg.conf ``` by commenting option ```option check_signature 1```

1am, 2016-02-24:
Thank you @mantas-p for the solution. I've bumped into this issue and your post helped me out.

just-a-coder, 2018-05-19:
I am experiencing the same issue with a device I just ordered from Amazon. 
I did not build my firmware from source.

```
root@GL-USB150:~# opkg update
Downloading http://www.gl-inet.com/openwrt/ar71xx/2.25/packages/base/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_base.
Downloading http://www.gl-inet.com/openwrt/ar71xx/2.25/packages/base/Packages.sig.
Signature check failed.
Remove wrong Signature file.
Downloading http://www.gl-inet.com/openwrt/ar71xx/2.25/packages/luci/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_luci.
Downloading http://www.gl-inet.com/openwrt/ar71xx/2.25/packages/luci/Packages.sig.
Signature check failed.
Remove wrong Signature file.
```

valinskas, 2021-09-27:
Closing issue, 5y+ since last activity.

closed 2021-09-27 by valinskas