8devices/carambola2 #65: GPIO 18 and 19 being blocked by I2C module

arthurbenemann, 2016-07-29:
Not sure if this is the correct place, please redirect me if not.

I need to control the GPIO pins 18 and 19, but they look to be being used by the i2c module.
```
root@carambola2:/# cat /sys/kernel/debug/gpio
GPIOs 0-29, ath79:
 gpio-18  (sda                 ) in  lo
 gpio-19  (scl                 ) in  lo
```

Also there doesn't seem to be any GPIO modules loaded on the kernel:
```
root@carambola2:/# lsmod | grep i2c
root@carambola2:/# lsmod
Module                  Size  Used by    Tainted: G
usb_storage            32640  0
ath79_wdt               2224  1
ledtrig_netdev          3024  0
nf_nat_irc               784  0
nf_nat_ftp              1040  0
nf_conntrack_irc        2624  1 nf_nat_irc
nf_conntrack_ftp        4912  1 nf_nat_ftp
ipt_MASQUERADE          1056  1
iptable_nat             1776  1
xt_nat                   992  0
nf_nat_ipv4             2528  1 iptable_nat
nf_nat                 10112  6 nf_nat_irc,nf_nat_ftp,ipt_MASQUERADE,iptable_nat,xt_nat,nf_nat_ipv4
pppoe                   7760  0
xt_conntrack            2080  3
xt_CT                   2320  0
iptable_raw              560  1
xt_state                 608  0
nf_conntrack_ipv4       5056  4
nf_defrag_ipv4           656  1 nf_conntrack_ipv4
nf_conntrack           41024 12 nf_nat_irc,nf_nat_ftp,nf_conntrack_irc,nf_conntrack_ftp,ipt_MASQUERADE,iptable_nat,nf_nat_ipv4,nf_nat,xt_conntrack,xt_CT,xt_state,nf_conntrack_ipv4
ehci_hcd               34128  0
sd_mod                 24560  0
pppox                   1136  1 pppoe
ipt_REJECT              1728  2
xt_TCPMSS               2560  0
xt_LOG                  9344  0
xt_comment               400  0
xt_multiport            1104  0
xt_mac                   528  0
xt_limit                 944  1
iptable_mangle           832  1
iptable_filter           592  1
ip_tables               9584  4 iptable_nat,iptable_raw,iptable_mangle,iptable_filter
xt_tcpudp               1584  4
x_tables               10160 17 ipt_MASQUERADE,xt_nat,xt_conntrack,xt_CT,iptable_raw,xt_state,ipt_REJECT,xt_TCPMSS,xt_LOG,xt_comment,xt_multiport,xt_mac,xt_limit,iptable_mangle,iptable_filter,ip_tables,xt_tcpudp
msdos                   5840  0
ppp_async               6128  0
ppp_generic            19488  3 pppoe,pppox,ppp_async
slhc                    4096  1 ppp_generic
vfat                    7984  0
fat                    41696  2 msdos,vfat
ath9k                  88816  0
ath9k_common            1184  1 ath9k
ath9k_hw              352016  2 ath9k,ath9k_common
ath                    14464  3 ath9k,ath9k_common,ath9k_hw
nls_utf8                 784  0
nls_iso8859_1           2816  0
nls_cp437               4352  0
mac80211              290032  1 ath9k
lib80211_crypt_tkip     6368  0
lib80211_crypt_ccmp     3504  0
lib80211_crypt_wep      1920  0
lib80211                2592  3 lib80211_crypt_tkip,lib80211_crypt_ccmp,lib80211_crypt_wep
usbcore               101152  2 usb_storage,ehci_hcd
usb_common               480  1 usbcore
scsi_mod               71440  2 usb_storage,sd_mod
nls_base                4704  6 vfat,fat,nls_utf8,nls_iso8859_1,nls_cp437,usbcore
crc_itu_t                944  0
crc_ccitt                944  1 ppp_async
eeprom_93cx6            1632  0
cfg80211              166016  3 ath9k,ath,mac80211
compat                  4832  9 ath9k,ath9k_common,ath9k_hw,mac80211,lib80211_crypt_tkip,lib80211_crypt_ccmp,lib80211_crypt_wep,lib80211,cfg80211
arc4                    1232  2
crypto_blkcipher        9664  1 arc4
aead                    4000  0
crypto_hash             8320  0
```

Do I need a new hardware design? Or is there a way to disable the i2c module?

The datasheet did mention this pins as being hassle free: ```Free GPIO pins: If you have some hoby project, it is advised to use GPIO pins 18, 19, 20, 21, 22 and 23 without any
worries. These pins are not used during the booting process. ````

arthurbenemann, 2016-07-29:
Maybe this commit has something to do with it https://github.com/8devices/carambola2/commit/862f420e4a8ec5455ce06a730566d8dd7a1b6d35

mantas-p, 2016-07-29:
Hi,

You can disable i2c module by editing ```/etc/modules.d/50-i2c-gpio-custom``` file:

```
i2c-dev
#i2c-gpio-custom bus0=0,18,19
```

arthurbenemann, 2016-07-29:
@mantas-p Thanks for replying.

I tried that, but the module was not in the modules.d folder. Note that I have removed the `leds_gpio` module previously trying to free up the GPIOS, but it had no effect on GPIO 18 and 19.

```
root@carambola2:~# ls /etc/modules.d/
02-crypto-hash     20-scsi-core       27-ath9k-common    40-usb2
03-crypto-manager  20-usb-core        28-ath9k           41-ipt-conntrack
09-crypto-arc4     21-lib80211        30-fs-vfat         41-pppoe
20-cfg80211        21-mac80211        30-ppp             42-ipt-nat
20-eeprom-93cx6    25-nls-cp437       40-fs-msdos        45-ipt-nathelper
20-lib-crc-ccitt   25-nls-iso8859-1   40-ipt-core        50-ledtrig-netdev
20-lib-crc-itu-t   25-nls-utf8        40-pppox           50-wdt-ath79
20-nls-base        26-ath             40-scsi-core       60-usb-storage
root@carambola2:~#
```

I'm still using the factory firmware. here is the version information:
```
root@carambola2:~# cat /proc/version
Linux version 3.7.9 (zilvinas@LX-buildd3) (gcc version 4.7.3 20121205 (prerelease) (Linaro GCC 4.7-2012.12) ) #8 Fri Jan 3 14:43:51 EET 2014
```

arthurbenemann, 2016-08-02:
@mantas-p Building from the source and flashing the device fixed the issue. IT was likely an issue with an older factory version.

closed 2016-08-02 by arthurbenemann