8devices/carambola2 #66: Recovering stack trace on dumped core

mpbn4, 2016-09-21:
Hello, 

I know it is not an issue, but I didn't find any other place to ask this question directly.

Could you help me to recover a stack trace for MIPS architecture same as specified in:

http://stackoverflow.com/questions/9809810/gdb-corrupted-stack-frame-how-to-debug

for x86? I have a core dump with same problem.

I have a cross compilation environment on my PC and the core dump comes from the target. 

When I modify 

    set $pc = *(void **)$esp

to:

    set $pc = *(void **)$sp.

I get "you can't do that without a process to debug". I would like to know if it is possible to recover this stack trace the way described in the above topic.

mantas-p, 2016-09-26:
Hi,

You can use gdbserver on Carambola2 and connect to it with gdb running on PC to have live debug session. Then this trick might work. You can even try to install gdb package on Carambola2, though you may run out of RAM or flash space doing so.

Another idea is to modify coredump file itself: http://stackoverflow.com/questions/7611082/linuxmips-temporarily-change-register-contents-when-viewing-core-dump didn't try that, but may work for you.

valinskas, 2021-09-27:
Closing issue, 5y+ since last activity.

closed 2021-09-27 by valinskas