8devices/carambola2 #69: Unique identification of Carambola2

1am, 2017-01-12:
Hello,

I would like to ask what is the recommended unique value to be used for device identification?
Relying on MAC address (which I've done so far) comes with a risk of running into duplicate values quite soon. 

Since I need a value which will also survive flashing from U-Boot I'm looking for something which would be tied to the hardware itself. I've stumbled upon a related OpenWRT topic: [Unique ID for TL-WR703N running openwrt](https://forum.openwrt.org/viewtopic.php?pid=234259#p234259). The value survived full reflashing.  

Given your experience with the Artheos and OpenWRT would you possibly advise for using the UUID value as a unique identifier with lower chance of collision?

bome, 2017-01-12:
There is a part of the flash memory which is not overwritten when flashing a new firmware. You can store your own data (e.g. a UID) there.

However, I wonder why MAC addresses will not be unique soon?

valinskas, 2017-01-12:
There  is a better way i think. According AR9331 datasheet: `Additional features include on-chip one-time programmable (OTP) memory`. Assuming there is enough information how actually write to OTP you can generate uniq IDs that survive anything. It's one time write and there you go - you need to worry how to generate a truly unique IDs.

Just an idea.

1am, 2017-01-12:
Thank you for your replies.

@bome This highly depends on how 8devices assigns the MAC addresses. Since each module has 3 MAC addresses (eth0, eth1, wlan0) the space would be limited to (65536^2)/3 (first 2 bytes are vendor identifiers so we're left with only 4 byte space). If 8devices assigns them sequentially then it's ~1.4M possible values which of course is a lot but still finite. If it's not assigned sequentially but randomly then possible duplicates are possible. Some vendors advise not to use MAC address as unique identifier for their modules/chips because of this reason. I wanted to start a discussion on 8devices approach by asking my question.

@valinskas This seems very clever. Do you maybe have a reference of accessing OTP for read-once/write from OpenWRT level?

bome, 2017-01-12:
I'd assume that 8devices would then just purchase another block of MAC addresses? AFAIK, they're quie cheap, too. Btw, you can also purchase blocks of MAC addresses yourself and overwrite them in the flash memory.

I have not heard of OTP memory, I use the standard flash memory: there is an area which OpenWRT is using for the MAC addresses and WiFi calibration data. You need to use a modified kernel to be able to get write access to that flash area. I store my data in that same block, in an area unused by OpenWRT.

pepe2k, 2017-01-12:
> You need to use a modified kernel to be able to get write access to that flash area.

Or just use something like https://github.com/jclehner/mtd-rw

Cheers,
Piotr

valinskas, 2017-01-12:
```
# insmod mtd-rw.ko i_want_a_brick=1
```

;)

bome, 2017-01-12:
@valinskas brilliant! that'll simplify our burn-in process a lot!

valinskas, 2017-01-12:
@1am: look here:
 https://github.com/zioproto/SDK.UBNT.v5.5/blob/master/target/linux/ubnt/files/arch/mips/atheros/otp.c 

Or such similar. Need to look for 9331 stuff, this can be more 934x related.

mantas-p, 2017-01-13:
@1am 8devices assigns MAC addresses sequentially and all modules sold must have unique address. We have no intention to change that in future, so it can be safely used as unique identifier.

1am, 2017-01-13:
Thank you for clarifying @mantas-p

closed 2017-01-13 by 1am