8devices/carambola2 #70: U-Boot errors

1am, 2017-02-02:
Hello,

Recently I've experienced a few strange issues with how UART functions on Carambola2 devices. I'm still investigating the PCB's themselves but for example on one of the devices which have Carambola2 mounted the UART output from C2 was the following:

```
U-Boot 1.1.4-g682743ef (Jan 20 2016 - 14:54:16)

=====================================
Caraboot v2.3 (AR9331) U-boot
http://www.8devices.com/                                                       IAL 64 MB----------------------
K@@
```

Later on it's frozen and nothing more happens after the corrupted message.
Is there something I could do to fix this or is this some physical error in internal memory?

mantas-p, 2017-02-03:
Hi,

Is message exactly the same every time or it changes between reboots? What UART adapter you are using to connect to Carambola2 module? Maybe you can check with oscilloscope/logic analyzer if baud rate is correct (Tx/Rx matches).

Similar symptoms can also be caused by defective device with corrupted boot-loader image or bad DRAM. Do you have problems with more than one sample?

1am, 2017-02-03:
Hi @mantas-p 

This issue is repeatable, occurs in more ~4 devices I think (each has different kind UART output, but not the expected one). Baud rate, etc. is ok as it passes on other boards we're mounting on and in general it was always the same.

In general by now we have quite a few defective modules (with different issues) but I still don't know where and how to ship them for replacement. support@8devices.com hasn't provided me with such answer when I asked. Is it possible that you discuss with them and we arrange for replacement?

mantas-p, 2017-02-03:
Looks like a HW defect to me. I'll ping support, sorry for delay.

valinskas, 2021-09-27:
Closing issue, 4y+ since last activity.

closed 2021-09-27 by valinskas