8devices/carambola2 #71: Overclocking carambola2

pbn4, 2017-02-27:
Hello, 

What is the easiest way to overclock the carambola2? I took a look into Caraboot source coude and I can see that there is an option to modify PLL clock for CPU/DDR/AHB, but that would require building and flashing the bootloader. Is there any way to do this without reflashing the bootloader? 

Also, from the values posted below, are all of those available for your module? 

```
/*
 * PLL Config for different CPU/DDR/AHB frequencies
 */
#define CFG_PLL_720_600_200	0x01
#define CFG_PLL_720_680_240	0x02
#define CFG_PLL_720_600_240	0x03
#define CFG_PLL_680_680_226	0x04
#define CFG_PLL_720_600_300	0x05
#define CFG_PLL_400_400_200	0x06
#define CFG_PLL_560_450_220	0x07
#define CFG_PLL_550_400_200	0x08
#define CFG_PLL_550_600_200	0x09
#define CFG_PLL_600_600_200	0x0a
#define CFG_PLL_750_400_250	0x0b
#define CFG_PLL_800_400_266	0x0c
#define CFG_PLL_750_667_250	0x0d
#define CFG_PLL_800_600_266	0x0e
#define CFG_PLL_800_667_266	0x0f
#define CFG_PLL_810_700_270	0x10
#define CFG_PLL_810_666_270	0x11
#define CFG_PLL_775_650_258	0x12
#define CFG_PLL_650_400_200	0x13
#define CFG_PLL_650_600_200	0x14
```

pepe2k, 2017-02-27:
> Is there any way to do this without reflashing the bootloader?

I don't think so. Clocks are setup during low level stage.
Maybe this will be interesting for you: https://github.com/pepe2k/u-boot_mod

Cheers,
Piotr

valinskas, 2021-09-27:
Closing issue, 4y+ since last activity.

closed 2021-09-27 by valinskas