8devices/carambola2 #72: Building - recipe for target 'world' failed

jtelocken, 2017-10-02:
Hello everyone,
I have the same issue.
Has anyone been able to solve this case? I'm using OpenSuse Leap 42.3. I've tried practically all the tips found in Web, about this problem and I can not solve it. I'll be grateful for any tips.

make[1] world
make[2] tools/install
make[2] package/cleanup
make[3] -C tools/flock compile
make[3] -C tools/flock install
make[3] -C tools/tar compile
make -r world: build failed. Please re-run make with -j1 V=s to see what's going on
/home/jonas/buildroot_desenv/3buildroot_novo/carambola2/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 1

Thanks for now.

mantas-p, 2017-10-02:
Hello,

OpenWRT build is best supported on Debian/Ubuntu distributions. I've just tested on Ubuntu 16.04.3 LTS and it builds just fine. I suggest to use virtual machine or docker image with Debian or Ubuntu for building.
If you still want to build on OpenSUSE, try building with debug turned on, so you can find where it fails:

```make -j1 V=s```

valinskas, 2017-10-02:
/home/jonas/buildroot_desenv/3buildroot_novo/carambola2/include/toplevel.mk:192: recipe for target 'world' failed

It seems like a hardcoded path somewhere ?

mantas-p, 2017-10-02:
@valinskas no, it's based on build dir location.

valinskas, 2017-10-02:
It works here:

```
$ git clone https://github.com/8devices/carambola2.git
$ cd carambola2
$ ./build.sh carambola2
 ... lots of output snipped
Installing package 'libssh' from management
Installing package 'libnetconf2' from management
Installing package 'libyang' from management
Installing package 'shtool' from management
Installing all packages from feed targets.
Collecting package info: done
#
# configuration written to .config
#
 make[1] world
 make[2] tools/install
 make[2] package/cleanup
 make[3] -C tools/flock compile
 make[3] -C tools/flock install
 make[3] -C tools/tar compile
 make[3] -C tools/tar install
 make[3] -C tools/patch compile
 make[3] -C tools/patch install
 make[3] -C tools/ccache compile
 make[3] -C tools/ccache install
 make[3] -C tools/sed compile
 make[3] -C tools/m4 compile
... continues without any apparent problem
```

jtelocken, 2017-10-02:
Hi Folks,

Thank you very much for your quickly answer. Certainly its bound with Opensue. I will test on Ubuntu.

Thanks again.

nishaero, 2017-10-26:
Hi jtelocken,
I tried to build from the latest Ubuntu 17.10 but it fails to build as well, seems like a configuration issue of some sort it would be great if someone could fix it. Im trying to build for Centipede device.

Thank You
Nishant

valinskas, 2021-09-27:
Closing issue, ≈3y  since last activity.

closed 2021-09-27 by valinskas