8devices/carambola2 #73: Testing Wi-Fi

pbn4, 2017-12-22:
Hello,

How one can configure Wi-Fi hardware to broadcast pseudo-random singal in various standards e.g. 802.11b,  802.11g, 802.11n? Working with OpenWRT/LEDE. So far I went through OpenWRT/LEDE docs and found nothing about configuring directly the transceiver.

valinskas, 2018-01-08:
Hello,

There is no standard way to send a packet at given modulation (11b/11g/11a/11n) from user-space. You will need to hack wireless driver directly. Say by adding some ioctl() handler and doing all the magic in there. For example: in ${KERNEL}/drivers/net/wireless/ath/ath9k/xmit.c file for ath9k driver is the code that is do transmit part, in ath_buf_set_rate() function TX rate is being selected and etc.

I hope it helps.

pbn4, 2018-01-09:
Thank you for your feedback. Could you provide us the utilities used for your certification procedure? We only need to validate that it works correctly in front of our authorization authority.

mdmannan179, 2018-04-18:
1c:a6:f7:21:28:e8

valinskas, 2021-09-27:
Closing issue, 2y+ since last activity.

closed 2021-09-27 by valinskas