8devices/carambola2 #74: Differences between carambola2 and openwrt?

erichiggins, 2019-03-08:
Hello,

I'm supporting a project which uses the Carambola2 board and I wanted to understand the steps to upgrade a few dependencies that ship with the firmware. Looking through this repo, it appears to be a fork of an older version of OpenWRT, plus some custom changes. So, I'd like to know three things:

1. Which version/release of OpenWRT is this repo currently based off of?
2. How much do the two differ? (Is there a diff available, or instructions on how to apply the Carambola2-specific changes?)
3. What is the recommended upgrade path for the included dependencies? (In this case, I'm hoping to upgrade the version of NodeJS to something more recent than 2014.)

Thank you!

mantas-p, 2019-03-13:
Hi,

 
>     1. Which version/release of OpenWRT is this repo currently based off of?
It is based on this commit
https://github.com/8devices/openwrt-8devices/commit/cda52dcb329e02674175b60426be26cd813294a6 which is around chaos_calmer release, but uses master branch

>     2. How much do the two differ? (Is there a diff available, or instructions on how to apply the Carambola2-specific changes?)

checkout and do this
```git diff cda52dcb329e02674175b60426be26cd813294a6..master```

> 
>     3. What is the recommended upgrade path for the included dependencies? (In this case, I'm hoping to upgrade the version of NodeJS to something more recent than 2014.)

Actually you can use upstream OpenWRT with Carambola2 - hardware is fully supported there.

erichiggins, 2019-03-13:
Great, thank you!

closed 2019-03-13 by erichiggins